marlinarnz/eVCU #2: Interrupt attach issue

jetpax, 2023-02-19:
Thank you for your elegantly crafted project which I think has a lot of potential!

I'm trying to run the Switch example and seem to have issues with the interrupt not firing.

I'm using an ESP32-S3 Devkit C , using the boot button on ```GPIO_NUM_0```

On boot I get this message in the console

```C++
ESP-ROM:esp32s3-20210327

Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
E (85) gpio: esp_ipc_call_blocking failed (0x103)
[    85][E][esp32-hal-gpio.c:175] __attachInterruptFunc
===== Starting up the devices =====

Started DeviceListener
Registering parameter with ID 0

===== Starting the test =====
```

I do get 1 warning during compile, but it does not seem related

```C++
src/DeviceSPI.cpp: In static member function 'static spi_device_interface_config_t DeviceSPI::getSlaveConfig(int, uint32_t, uint16_t)':
src/DeviceSPI.cpp:344:3: warning: narrowing conversion of 'speed' from 'uint32_t' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
   };
```

Any thoughts?

marlinarnz, 2023-02-19:
Hi @jetpax , thanks for having a look at this project!

I am not sure, whether `GPIO_NUM_0` is interrupt-compatible, but I have never worked with the ESP S3. This is a good compatibility test, as some peripherals are implemented differently here. However, I think you should first try to use any other GPIO and connect it to GND with a jumper to emulate a button pressed (if passing `INPUT_PULLUP` as pin mode to the `Switch` constructor). It looks like your interrupt has been blocked by the boot process.

Moreover, I made a little bugfix for code stability, combined with a little `Device` class interface change. That shouldn't bother, but I would recommend pulling it. It also features Arduino-style library characteristics, even though it is not listed yet. The Switch example works for my standard DevKit v1.

jetpax, 2023-02-20:
Thanks, I already had pulled your latest interface change, much cleaner!

I tried with another pin, but with same result unfirtunately

Using my environment (Platformio 5.3, so ESP-IDF 4.4.3 with ArduinoEspressif 2.0.6) I also tried the default Arduino GPIO interrupt sketch, and that worked fine (even with GPIO0)

```
#include <Arduino.h>

struct Button {
    const uint8_t PIN;
    uint32_t numberKeyPresses;
    bool pressed;
};

Button button1 = {0, 0, false};
Button button2 = {18, 0, false};

void ARDUINO_ISR_ATTR isr(void* arg) {
    Button* s = static_cast<Button*>(arg);
    s->numberKeyPresses += 1;
    s->pressed = true;
}

void ARDUINO_ISR_ATTR isr() {
    button2.numberKeyPresses += 1;
    button2.pressed = true;
}

void setup() {
    Serial.begin(115200);
    pinMode(button1.PIN, INPUT_PULLUP);
    attachInterruptArg(button1.PIN, isr, &button1, FALLING);
    pinMode(button2.PIN, INPUT_PULLUP);
    attachInterrupt(button2.PIN, isr, FALLING);
}

void loop() {
    if (button1.pressed) {
        Serial.printf("Button 1 has been pressed %u times\n", button1.numberKeyPresses);
        button1.pressed = false;
    }
    if (button2.pressed) {
        Serial.printf("Button 2 has been pressed %u times\n", button2.numberKeyPresses);
        button2.pressed = false;
    }
    static uint32_t lastMillis = 0;
    if (millis() - lastMillis > 10000) {
      lastMillis = millis();
      detachInterrupt(button1.PIN);
    }
}
```

Couldn't find anything really similar out there, except maybe this https://esp32.com/viewtopic.php?t=26062
???

marlinarnz, 2023-02-20:
Thanks for trying! I think it could be a trivial debounce issue, as I had never tried it with a button before. Please try pushing the button a bit longer as well. And I just fixed another issue: The parameter connected to the switch position was initialised with 0, even if its physical state is 1. Hence, I added an update in the `Switch`'s `begin()` function to start with the physical value. I hope that works.

Otherwise, it must be an FreeRTOS/multi-threading issue, as the above example works for you. I hope we don't have to look into that...

jetpax, 2023-02-20:
In fact, although the simple Arduino interrupt example above did work, the more sophisticated FunctionalInterrupt.ino  example for ESP32 does not (with the ESP32-S3), and indeed it errors on start with the same sort of failure message I was seeing with eVCU.

Turns out it was a problem with attaching the interrupt in the pin constructor as hinted above by ESP_Sprite.

The fix for the FunctionalInterrupt.ino example is simply to attach the interrupt when the device is started, not when it is constructed, ie 

```C++
#include <Arduino.h>
#include <FunctionalInterrupt.h>

#define BUTTON1 0
#define BUTTON2 18

class Button
{
public:
	Button(uint8_t reqPin) : PIN(reqPin){
		pinMode(PIN, INPUT_PULLUP);
        // attaching the interrupt here causes ESP32-S3 to crash with GPIO isr service already installed error
	// attachInterrupt(PIN, std::bind(&Button::isr,this), FALLING);
	};

    void begin(){
        attachInterrupt(PIN, std::bind(&Button::isr,this), FALLING);
    }

	~Button() {
		detachInterrupt(PIN);
	}

	void ARDUINO_ISR_ATTR isr() {
		numberKeyPresses += 1;
		pressed = true;
	}

	void checkPressed() {
		if (pressed) {
			Serial.printf("Button on pin %u has been pressed %u times\n", PIN, numberKeyPresses);
			pressed = false;
		}
	}

private:
	const uint8_t PIN;
    volatile uint32_t numberKeyPresses;
    volatile bool pressed;
};

Button button1(BUTTON1);
Button button2(BUTTON2);

void setup() {
    Serial.begin(115200);
    button1.begin();
    button2.begin();
}

void loop() {
	button1.checkPressed();
	button2.checkPressed();
}
```
I fixed this in eVCU by modifying the DevicePin class, and it appears to work fine, even using the boot pin!

```C++
/** The constructor does not attach the interrupt, as this causes issues with some devices (eg ESP32-S3)
 *  Instead, this should be done when the device is started, eg in Switch::begin()
 *  @param pController: VehicleController instance passed to Device
 *  @param pin: GPIO pin for the interrupt
 *  @param debounce: minimum millis between two interrupt yields
 *  @param inputMode: INPUT, INPUT_PULLUP, INPUT_PULLDOWN
 *  @param interruptMode: at pin voltage LOW, CHANGE, RISING, FALLING
 */
DevicePin::DevicePin(VehicleController* pController, uint8_t pin, int debounce, int inputMode, int interruptMode)
  : Device(pController), m_taskHandleOnPinInterrupt(NULL),
    m_pin(pin), m_debounce(debounce), m_interruptMode(interruptMode)
{
  pinMode(this->m_pin, inputMode);
}

void DevicePin::attachInt()
{
  attachInterruptArg(this->m_pin, isr, &m_taskHandleOnPinInterrupt, CHANGE);
}

/** The destructor deletes tasks and detaches the interrupt.
 */
DevicePin::~DevicePin()
{
  if(m_taskHandleOnPinInterrupt != NULL){
    detachInterrupt(this->m_pin);
    vTaskDelete(m_taskHandleOnPinInterrupt);
    m_taskHandleOnPinInterrupt = NULL;
  }
}
```
and modified Switch.cpp accordingly

```C++
void Switch::begin()
{
  this->attachInt();
  this->startTasks(4096, 8192);
}

```

Not sure if/how you want to fix this, but hope this is of some use.

closed 2023-02-20 by jetpax

marlinarnz, 2023-02-21:
Thanks for figuring it out. I implemented and validated your suggestion of attaching in `begin()` with `attachISR()`. I updated the documentation accordingly.