marlinarnz/eVCU #6: Pedal Example not working

jetpax, 2023-02-22:
Pedal example is not working on ESP32-S3.

I tried adding a call to ```attachISR()``` in ```Pedal::begin()```, but this had no effect.

Basic Arduino AnalogRead.ino works fine.

marlinarnz, 2023-02-24:
I found some smaller issues, but the problematic one is that the ESP32 does not support interrupts on an analog value change. Hence, I added a new base class that simply runs one function in a loop, given an interval. This is polling, in contrast to event-driven designs, but it actually makes sense for a throttle. Polling every 10ms has a better performance than interrupting every 1ms due to the noise of the ADC or poti.

c616ddab18b58bf4f26447d62cfc448cc19d1e39

closed 2023-02-24 by marlinarnz

jetpax, 2023-02-25:
A periodic 'device'  was sorely needed!

Unfortunately moving the pin mode setting out of the contructor and into the attachISR() method breaks ESP32-S3.

Moving it back, both Switch and Pedal examples work great for me, thanks Marlin!

marlinarnz, 2023-02-25:
I ordered an S3 now, too 😆 And I moved it back to the constructor. Also for the new `DevicePinMulti` template.

jetpax, 2023-02-25:
😆 The S3 is awesome, especially the inbuilt debugger which works pretty well in Platformio. All you need is an extra USB cable and you can develop/debug anywhere.

[image]

IMO the DevkitM is better than the DevkitC, if you can get hold of one; the former uses the smaller module but ironically has two extra pins.  Plus, both the ESP32-S3-Mini module layout and the DevkitM PCB layout are more sane which can only help the ADC...

[image]