marlinarnz/eVCU #7: CAN bus recovery error

marlinarnz, 2023-04-16:
When connected to only one other unstable CAN bus node, the ESP32 crashes and continuously reboots with this message:

```
...
Error sending CAN message: Try to start the CAN driver
Info: Started CAN bus recovery
Error sending CAN message: Try to start the CAN driver
Error sending CAN message: Try to start the CAN driver
Info: Started CAN bus recovery

assert failed: twai_handle_tx_buffer_frame twai.c:183 (p_twai_obj->tx_msg_count >= 0)

Backtrace: 0x4008353d:0x3ffbf14c |<-CORRUPTED
```

Potentially solve it by emptying the TX messages before restarting/recovering.

marlinarnz, 2023-06-18:
This issue seems to be common without being properly fixed at the moment:
https://github.com/espressif/esp-idf/issues/9697
https://github.com/espressif/esp-idf/issues/9169

In this case here, it seems to occur in conditions with unstable voltage supply and message send cycles of 10ms or less.

marlinarnz, 2024-01-10:
Occurs when bus-off because of too many transmit fails
``` 
Error sending CAN message: wrong state
Info: Started CAN bus recovery
Info: Waiting for CAN bus to recover
assert failed: twai_handle_tx_buffer_frame twai.c:183 (p_twai_obj->tx_msg_count >= 0)
```

Trying to solve it by setting TWAI config parameters in `C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\tools\sdk\esp32\sdkconfig`:

``` 
#
# TWAI configuration
#
# CONFIG_TWAI_ISR_IN_IRAM is not set
CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC=y
# CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=y --> disabled as crash fix
CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID=y
CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT=y
# CONFIG_TWAI_ERRATA_FIX_LISTEN_ONLY_DOM is not set
# end of TWAI configuration
```

italocjs, 2024-07-31:
Any updates on this?

marlinarnz, 2024-08-01:
The config bugfix above works seamlessly for the ESP32. Will test it with the ESP32-S3 soon.